NModbus/NModbus #112: Return 0 problem

jot4p, 2022-07-07:
Hi, 

I use the lib to read some modbus devices (with delay between readins with 10ms to 500ms) and sometimes (many times in some devices like schneider) the result is 0 , but if I try again 10x or 20x the value is correct (0 0 0 0 0 0 correct 0 0 0 0 0 0 0 correct 0 0 0 0 0 0 0 0 0 etc..) ...  

If I use external software, with 1ms, 5ms or 100 milliseconds (example) of delay between readings the results are always ok (never 0). 

Any Idea where is the error ?   Nmodbus 3.0.62

MPapst, 2023-08-25:
This may be an issue with the schneider device.
Is that related to TCP or serial connections?

To dig a bit deeper it would be good if you could get a capture of the traffic on the line (e.g. with Wireshark).

rquackenbush, 2023-10-31:
This issue has been stale for a long time, so I'm closing it for now. If you're still having this issue, please let me know.

closed 2023-10-31 by rquackenbush