NModbus/NModbus #117: Modbus TCP cannot run？

183demao, 2022-09-29:
[image]
I tried to run Modbus TCP mode, but it got stuck at the breakpoint，

When you trace the code, it doesn't go any further than this line

[image]

Can you help me ? thank you

MPapst, 2022-09-29:
The AccepTcpClient method waits for a client to connect. Thats expected behavior.

183demao, 2022-09-29:
I recorded the picture. I don't know if I can see it
[image]

Server is Runnning ，but client ReadInputs No further execution

183demao, 2022-09-29:
Thank you very much