NModbus/NModbus #118: Make *Request and *Response classes public

M3lun4, 2022-11-11:
In NModbus4 these classes where public. 

See: https://github.com/NModbus4/NModbus4/commit/aa654ca316796108afd743140d312a27bb996a7a

A public access modifier would help to achieve something like this in a Unit Test:

```
var message = (ReadHoldingInputRegistersRequest)this.MessageReceivedBySlave;
Assert.AreEqual(5, message.StartAddress);
```

rquackenbush, 2022-11-17:
I’m planning on making basically everything public / non-sealed in the next major version of NModbus. That said, I’ll see if I can crank out a version of the current code base that marks those as public.

rquackenbush, 2023-08-21:
The request and response classes have been made public and are available as of version 3.0.80 of the nuget package.

closed 2023-08-21 by rquackenbush