NModbus/NModbus #119: How to execute writemultiplecoilsasync?

gotnetdude, 2022-11-16:
I could use some help getting the WriteMultipleCoilsAsync task to run. I'm getting the error message Cannot assign void to an implicitly-typed variable on the Task.Run statement. I need to capture the response message from WriteMultipleCoilsAsync.  I have been unable to find any examples or documentation searching the web.  

Here some example code:
```
private async void btnWriteMultipleCoils_Click(object? sender, EventArgs e)
    {
        string discretes = "true,false,true";
        bool[] diarray = Array.ConvertAll(discretes.Split(','), bool.Parse);
        var response = await Task.Run(() => master.WriteMultipleCoilsAsync(0, diarray));
        await Task.CompletedTask;
    }
```

Here is the WriteMultipleCoilsAsync task:
```
public Task WriteMultipleCoilsAsync(ushort startAddress, bool[] data)
    {
        return WriteMultipleCoilsAsync(0, startAddress, data);
    }
```
Here is the WriteMultipleCoilsAsync task:
```
public Task WriteMultipleCoilsAsync(byte slaveAddress, ushort startAddress, bool[] data)
    {
        ValidateData("data", data, 1968);
        WriteMultipleCoilsRequest request = new WriteMultipleCoilsRequest(slaveAddress, startAddress, new DiscreteCollection(data));
        return PerformWriteRequestAsync<WriteMultipleCoilsResponse>(request);
    }
```

rquackenbush, 2022-11-17:
Can you do something like:

private async void btnWriteMultipleCoils_Click(object? sender, EventArgs e)
    {
        string discretes = "true,false,true";
        bool[] diarray = Array.ConvertAll(discretes.Split(','), bool.Parse);

        var response = await master.WriteMultipleCoilsAsync(0, diarray);
    }

gotnetdude, 2022-11-17:
Unfortunately no,  same error.

gotnetdude, 2022-11-17:
I tried the following, but still get the same error:

```
private async void btnWriteMultipleCoils_Click(object? sender, EventArgs e)
    {
        string discretes = "true,false,true";
        bool[] diarray = Array.ConvertAll(discretes.Split(','), bool.Parse);
        Task<IModbusMessage> response = await Task.Run(() => master.WriteMultipleCoilsAsync(0, diarray)); 
        await Task.CompletedTask;
    }
```
Do you think maybe changing the return type of the three methods WriteMultipleCoilsAsync(2 args), WriteMultipleCoilsAsync(3 args) and PerformWriteRequestAsync from Task to Task<WriteMultipleCoilsResponse> would make a difference.

gotnetdude, 2022-11-17:
I feel certain that the method` [ModbusMaster.PerformWriteRequestAsync<T>]` violates [The Guideline](https://devblogs.microsoft.com/pfxteam/should-i-expose-asynchronous-wrappers-for-synchronous-methods/) about not exposing asynchronous wrappers for synchronous methods, as well as the [CA2008](https://learn.microsoft.com/en-us/dotnet/fundamentals/code-analysis/quality-rules/ca2008) guideline.  This information was provided by Theodor Zoulias Senior Developer reached via Stack Overflow regarding the [Execute async await task WriteMultipleCoilsAsync issue](https://stackoverflow.com/questions/74468111/execute-async-await-task-writemultiplecoilsasync-issue).  Theo recommended the above changes.   Hopefully we can try it out and see if they work?  Can you please advise me on how we proceed with the proposed changes.

MPapst, 2023-01-21:
The Method `WriteMultipleCoilsAsync` on `IModbusMaster` returns a `Task`.
What are you expecting from it to return?

From modbus point of view, you are writing values to a different device and you are not expecting a response (either you are getting a modbus exception, there is no response to that modbus function code).

When only a Task is returned, you can await it and have no result (so you cannot assign an awaited Task, that is what the error means) or you can assign it to a variable of type Task and await that variable, which leaves you the possibility.

about that guideline: That may or may not be, but it does not affect what you are trying to achieve.

closed 2023-08-21 by rquackenbush