NModbus/NModbus #120: Support for broadcast messaging

cdrose, 2022-11-17:
Is there a way to implement broadcast messaging in NModbus?

A broadcast message is sent to node 0 and has no response from the slave devices. As far as I can see writes to node 0 are handled the same as any other message and wait for a response. I have tried to work around it by setting the the read timeout to 1, retries to zero and then catching the TimeoutException but the performance is not great. Is there a better approach?

rquackenbush, 2022-11-17:
I haven’t looked into this closely in the current implementation. I suspect that you’ve found the best workaround. 

I’m working on the next version of NModbus and am providing first class support for broadcast requests.

cdrose, 2022-11-17:
Is there a way to make use of the RTU message building functions so I can build the message and just send it directly over the serial port? Actually all I need is the CRC, assembling the rest of the message is trivial.

MondoKevin, 2023-06-07:
> I haven’t looked into this closely in the current implementation. I suspect that you’ve found the best workaround.
> 
> I’m working on the next version of NModbus and am providing first class support for broadcast requests.

Hi @rquackenbush. I am also looking at using the broadcast functionality. I was wondering if the new version with broadcast support would be available soon?

rquackenbush, 2023-07-28:
@MondoKevin - unfortunately it's still in progress. I'm actually looking for someone to sponsor some serial adapters for me to develop with. Any chance  you're interested? I'm looking for 3  FTDI RS-485 adapters.

rquackenbush, 2023-08-21:
I've ordered FTDI cables and will attempt to incorporate broadcast messaging into the 3.x version of NModbus.

MondoKevin, 2023-08-30:
> I've ordered FTDI cables and will attempt to incorporate broadcast messaging into the 3.x version of NModbus.

That's great! Thanks @rquackenbush. I look forward to seeing how it turns out.