NModbus/NModbus #123: 32-bit Holding Registers

BCooperCEESI, 2023-01-20:
I have a number of devices that have single 32-bit holding registers. I believe this is called "Enron Modbus". Looking through the code, it is clear it is not designed to handle 4 bytes of data from a single holding register and this throws errors. I am curious if anyone else has run into this, or knows of a solution to this? I am not opposed to modifying the source code to accommodate this option (and have started doing this), but time is sensitive for the project I am working on. Any suggestions would be appreciated.

rquackenbush, 2023-01-21:
You’re quite correct that you can’t fit a 32 bit integer in a 16 bit
holding register. How I’ve handled this in the past is to just read or
write two registers at the same time. You can then translate these to 32
bit integers using the BitConverter calls.

AgoraIOT, 2023-04-21:
These should have addressed by the Enron Modbus extensions.  I built the branch earlier, and some changes were recently merged into 3.0.74.  I will test 3.0.78 and see if it works.  My developers are telling me the latest release is not working.

@rquackenbush I'll get back to you soon on that.

jksware, 2023-05-08:
> You’re quite correct that you can’t fit a 32 bit integer in a 16 bit holding register. How I’ve handled this in the past is to just read or write two registers at the same time. You can then translate these to 32 bit integers using the BitConverter calls.

I've had issues when trying to write 32-bit holding registers using that trick: the other end is expecting 4 bytes of values but only one point (one register) in the message. Using the previous method, the number of points in the request message (TX) is 2, and the device throws an error response.

For instance, in Modbus/TCP, I want to write a 32 bit int to holding register 51900, where the device is expecting a 32-bit word, using two 16-bit registers with the current solution:
```
TX: 00 13 00 00 00 0B 01 10 CA BC 00 02 04 00 00 00 14
                                     --
                                 (# of points)                                      
RX: 00 13 00 00 00 03 01 90 03
```
The response from the device is (03) Illegal Data Value. Most probably this is due to an unexpected number of registers in the message: two, instead of one for a single 32-bit register (4 bytes) value.

What actually works for me:
```                                    
TX: 00 13 00 00 00 0B 01 10 CA BC 00 01 04 00 00 00 14
                                     --
                                 (# of points) 

RX: 00 13 00 00 00 06 01 10 CA BC 00 01
```

Notice this is a real world example, and the only difference between the successful and unsuccessful cases is the number of registers.

I've forked this repository (https://github.com/jksware/NModbus/tree/feature/64-bit-modbus) and made some change here and there to accept correctly writing 32-bit and reading 64-bit holding registers, where the device expects a `bytes/4` or `bytes/8` number of points in the message, and to be able to read 32 ASCII chars from a single register -- because a fixed number of bytes being expected per point. For making this possible, I've created additional Request and Response classes, in addition to specific methods to Read and Write 32- and 64-bit registers.

But before making a pull request, I'm wondering what the best solution to adopt this change would be:
1. Stick to having separate methods, and Request/Response classes for each 16, 32 and 64 bit sizes, and 256 (32 chars) via optional parameters , or
2. Adding bytesPerPoint, numberOfPoints optional parameters to Read/Write methods, and pass those down to the actual reading/writing methods, and don't create additional Request/Response classes, where possible.

Is there something else I'm missing?

closed 2023-08-21 by rquackenbush