NModbus/NModbus #124: Is this library thread safe?

UEPSs, 2023-02-14:
Is this library thread safe? I'm using it with serial rtu on .net 6 on windows.

rquackenbush, 2023-02-14:
No - it is not thread safe. I have run into situations where I needed it be thread safe, so I wrapped `IModbusMaster` with another class that provided locking. It was something like this:

```c#
LockingModbusMaster : IModbusMaster
{
   private readonly IModbusMaster inner;
   private readonly object lockObject = new object();
   private 
   
   public LockingModbusMaster(IModbusMaster inner)
   {
       this.inner = inner;
   }
  
   public ushort[] ReadHoldingRegisters(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
   {  
       lock(lockObject)
       {
            return inner.ReadHoldingRegisters(slaveAddress, startAddress, numberOfPoints);
       }
   }
}
```

Note that if you're using the async methods, a handy class comes from the https://github.com/StephenCleary/AsyncEx package:

```c#
LockingModbusMaster : IModbusMaster
{
   private readonly IModbusMaster inner;
   private readonly AsyncLock asyncLock = new AsyncLock();

   public LockingModbusMaster(IModbusMaster inner)
   {
       this.inner = inner;
   }

   public async Task<ushort[]> ReadHoldingRegistersAsync(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
   {  
       using (await _asyncLock.LockAsync())
       {
            return await inner.ReadHoldingRegisters(slaveAddress, startAddress, numberOfPoints);
       }
   }
}
```
**Note:** There is one more aspect of this that is both important and missing from my sketch above. For serial communications, there has to be quiet time between requests on the serial line. If I remember correctly, I used a stopwatch (or similar mechanism) to ensure that I had enough quiet time before sending the next command. Here is a sketch of how you could do it:

```c#
LockingModbusMaster : IModbusMaster
{
    private readonly IModbusMaster inner;
    private readonly AsyncLock asyncLock = new AsyncLock();
    private readonly Stopwatch stopwatch = Stopwatch.StartNew();
    private readonly TimeSpan minimumQuietTime;

    public LockingModbusMaster(IModbusMaster inner, TimeSpan minimumQuietTime)
    {
        this.inner = inner;
        this.minimumQuietTime = minimumQuietTime;
    }

    public async Task<ushort[]> ReadHoldingRegistersAsync(byte slaveAddress, ushort startAddress, ushort numberOfPoints)
    {  
        using (await _asyncLock.LockAsync())
        {
             await WaitAsync();

             var registers = await inner.ReadHoldingRegisters(slaveAddress, startAddress, numberOfPoints);

             stopwatch.Restart();
        }
   }

   private async Task WaitAsync()
   {
         var difference = stopwatch.Elapsed - minimumQuietTime;

         if (difference> TimeSpan.Zero)
         {
               await Task.Delay(difference);
         }
   }
}
```

I hope this helps.

LJN-hzleaper, 2023-03-07:
Thanks a lot.

closed 2023-08-21 by rquackenbush