NModbus/NModbus #132: Asynchronous Writing closes the SerialPort ?

Jisl2, 2023-05-24:
Hello,
I have a problem when I try to use the WriteSingleRegisterAsync.

Here is part of the code I wrote :
```
ModbusFactory mf_prog = new ModbusFactory();
IModbusSerialMaster master = mf_prog.CreateRtuMaster(MainWindow.port);

for (j = 0; j < add.Length; j++)
{
  if (typefile_array[i][0] == data_array[i][0])
  {
    if (add[i] != 0)
    {
      await master.WriteSingleRegisterAsync(1, add[i], values[i]);
    }
  }
  else
  {
    MessageBox.Show(String.Concat("Index difference : \nTypeFile : ", typefile_array[i][0], "  ;  DataFile : ", data_array[i][0]), "Error");
    return;
  }
  i++;
  MainWindow.OneDone(EventArgs.Empty);
}
  ```
When I try to write, an exception occurs telling me that the port is closed. It blocks at j=0, so no data is ever sent
I can also see that the port is opened until then;
And when I don't use an Asynchronous writing, the program does work.

Is there a way to slove this ? Maybe it's an obvious reason, but I'm not used to programming in c#

MPapst, 2023-05-29:
The serial Port can only be Süd exclusive. So if you want to send a second write, you must either close the serial port before opening it again or (I would recommend that) reuse it.

Jisl2, 2023-05-30:
Thanks for the answer, I'll try to put this as a condition, and as a related question is there a signal or a bool representing if the writing as been done successfully ? 
As an exemple : ('check' represents the var I'm talking about)
```
public void Writing()
{
    *Writing actions*
    if (var check == true)
    {
        return;
    }
    else
    {
        *throw Exception*
    }
}
```

MPapst, 2023-05-30:
You can use try...catch to capture exceptions. If there is no exception and you have awaited the async write task, the operation is done.

Jisl2, 2023-05-30:
Thanks, that actually solved part of my problem

One last thing that causes me an error is that in my program, I need to write several registers, but it seems that the program sends the next writing request without waiting for the response on the previous one. For example, I need to write something like 800 registers (in a loop like on my first message), but it takes my program only 7 or 8 seconds at 9600 baud before telling me it's finished, which is less than the best theoric time for request+response. The program waits for all the responses, but says that the writing is done while it's still waiting for responses. Can I implement something that waits for the reponsse of a transmission before sending another ?

MPapst, 2023-05-30:
If you are using await when calling the write method, it should wait before going into the next loop.
Have you tried debugging your code?

Jisl2, 2023-05-30:
Yes, and when I try debugging my code it tells me that it needs between 7 and 9ms to do a transmission (request+response), which seems quite fast to me. And no error is encountered by the program during the process.

closed 2023-08-21 by rquackenbush