NModbus/NModbus #133: Get the underlying byte array of a response

f-zl, 2023-05-29:
Thanks for the helpful library. I need to read a certain amount of holding registers from a modbus server. The response need to be interpreted as a whole (such as a C struct on a little endian machine), not separate registers. So it will be really convenient if functions like `IModbusMaster.ReadHoldingRegisters` can return a byte array, or a response object with accessor methods for the underlying byte array, like the `QModbusReponse` class and its `data()` method in Qt. Maybe the response classes can be public and methods that return those response objects can be added to `IModbusMaster`.

MPapst, 2023-05-29:
Per Modbus Specification, holding registers are 16 Bit values, which has the C# representation in ushort.
You can simply convert the response into a byte array:
```csharp
ushort[] holdings = await client.ReadHolding...
byte[] bytes = holdings.Select(register => BitConverter.GetBytes(register)).SelectMany(byte => byte).ToArray();
```

_Note_: untested, just to show how it could work.

rquackenbush, 2023-05-30:
I've used the same exact approach as @MPapst described. 

That said, I'm still working on the next major version of NModbus, and I might expose a modbus response object. Perhaps something along the lines of:

```c#
public interface IModbusResponse<T>
{
   byte UnitId { get;}

   MemorySpan<byte> Frame { get; }

   Value<T> Value { get; }
}
```

We would call it thusly:
```c#
IModbusResponse<ushort[]> response = await client.ReadHoldingRegistersAsync(1, 100, 10);

foreach(var register in response.Value)
{
   Console.WriteLine(register);
}
```

rquackenbush, 2023-07-28:
I'm going to close this as @MPapst has posted a usable workaround, and I'm planning on addressing this in NModbus.v4

closed 2023-07-28 by rquackenbush