NModbus/NModbus #139: Overriding Existing Function Services

Tyler-Duckworth, 2023-07-18:
Hi! 

I am in the process of porting some old code that uses NModbus4 to NModbus and I had some questions for the best way to modify the existing function services NModbus uses. In particular, our code modifies WriteSingleCoil and WriteMultipleCoil (using the `ModbusSlaveRequestReceived` event) to do some custom caching before letting NModbus execute the request. However, it seems like all of the classes related to these requests are `internal` in this new version. 

What would be the best way of overriding these function services to enable this functionality?

rquackenbush, 2023-07-18:
I made some suboptimal choices when I started refactoring NModbus when I
inherited the project. One of those was marking a bunch of things internal.

I see a few paths forward:
- we update the message classes to be public (PRs are welcome)
- wait for the next major version of NModbus (I don’t recommend this - I
don’t know when this effort will be done). It’ll be great! There is first
class support for plugging in custom function code implementations.
- move caching outside of the NModbus project (my knee jerk reaction says
this is a good approach). Perhaps put a façade in from of IModbusMaster, or
add some extension methods.

Tyler-Duckworth, 2023-07-27:
Sorry for the late response, but I'd definitely be interested in contributing to NModbus! What sort of requirements do you all look for in PRs and contributing in general?

rquackenbush, 2023-07-28:
@Tyler-Duckworth  - thanks for your interest in contributing! This is a small project, so I don't have a set of community / contribution guidelines. That said:
- Don't make breaking changes
- Add tests for new features
- Add comments where appropriate
- Don't change code unrelated to the focus of your PR

closed 2023-08-21 by rquackenbush