NModbus/NModbus #140: Recieve Response after action

adrianfernandeziturcemi, 2023-07-25:
Hello, i am testing this library for research purposes. Is there any way to receive the response of the slave after a write / read etc. ?

Thank you in advance.

MPapst, 2023-07-25:
Hi,

what kind of response are you expecting?
Most write function codes do not have a response. If you do not get an exception, the write operation has completed successfull.

rquackenbush, 2023-07-27:
For operations that return data, just the data is returned:

```c#
ushort[] holdingRegisters = modbusMaster.ReadMultipleHoldingRegisters(1, 10);
```

For operations that don't return data, nothing is returned:

```c#
modbusMaster.WriteMultipleRegisters(1, new ushort[0, 1, 2, 3, 4, 5]);
```

To handle errors, you have to wrap the call in a Try/Catch clause:

```c#
try
{
   modbusMaster.WriteMultipleRegisters(1, new ushort[0, 1, 2, 3, 4, 5]);
}
catch(Exception ex)
{
   //TODO: Log ex
}
```

rquackenbush, 2023-08-21:
If you have any additional questions, please let us know. Closing for now.

closed 2023-08-21 by rquackenbush