NModbus/NModbus #148: Can not read write address 65535 in holding registers

alfitec, 2023-09-17:
I have created simple server as in test code, and i can read:
slave1.DataStore.HoldingRegisters.ReadPoints(0, 65535 );
but i cant:
slave1.DataStore.HoldingRegisters.ReadPoints(65535, 1)

also with client I tried reading coils, I can read 65525..65534 but when i 
try 65525.65535 i get errorcode back.

Not realy important but we have one adress missing ... probably 0 base/1 base array error, also ushort might add some limitations for index/noumber of points :( ...

MPapst, 2023-09-17:
That's a limitation of the protocol definition and an interpretation issue.

You can either have the registers 0-based and read up to register 65534 (that's how it's done on the physical line) or 1-based and read up to register 65535, but this has to be a translation inside the application layer of your application.

You can not handle register 65535 (as number) because the protocol specification defines a 16-bit sized field for the register address.

alfitec, 2023-09-17:
[MODBUS Application Protocol Specification V1.1b3](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) says on page 15 for function 03 adress is 0x0000 .. 0xFFFF

Wikipedia says adresses are standard 1..9999   but de facto standard 1..65536 so after subtracting 1 we have adress 0..65535. This would be 16 bit address space.

I don't realy need this address,  and probably nobody does, I was just testing few lines of code and wanted to test first and last address...

MPapst, 2023-09-17:
You are right - and I am not wrong :)

On the ethernet line (talking about Modbus TCP) Register Address 1 translates to 0x0000. Which means 0xFFFF (decimal 65534) on the ethernet line translates to Register Address 65535.
This is an application layer topic, not a protocol.
This is because the Modbus message on the transport layer defines 2 Bytes as Address, which can only hold number between 0 and 65534 interpreting them as `unsigned int 16`.

So from a protocol perspective, its from 0 to 65534, from an application perspective you can go from 1 to 65535. The Modbus Specification says that on page 15: `In the PDU Registers are addressed starting at zero. Therefore registers numbered 1-16 are addressed as 0-15.`
I've seen a lot of applications and its always handled different... some are using as start and some 1 and you always need to try and error :)

For the library NModbus Library (for typed languages) its about data types. `ushort` is the best fit, but can not handle the decimal number 65535.

alfitec, 2023-09-17:
Yap, edit point is what i wanted to say... it should be reachable...
0xFFFF is 65535, 0x10000 is 65536 (quick check 65536 is even number so is 0x10000)

Not that it is super useful feature, but it also might be super easy to fix...
It even might be that only datastore should be initialised bigger or something like that...

there is unfortunate thing with limit beeing max ushort but there is relatively litle need to optimise library to use ushort where unt/uint could be used ... 

but I am not PC programmer, i am quite rusty (without any reference to rust which i dont realy know) with PC programming. I am more in PLC world.

rquackenbush, 2023-09-17:
I’ve gone back and forth on this a number of times with myself. ushort is a
good fit from my perspective as an implementer, but I am aware that
specifications for Modbus devices are available in both zero based and one
based addressed.

I’ve thought about coming up with a fancy struct that represents addresses
and allows either method. That might still be a viable option.

It could be something along the lines of:

public struct ModbusAddress
{
    public static ModbusAddress FromIndex(ushort index);

     public static ModbusAddress FromAddress(uint address);

     public ushort Index { get; }

     public uint Address { get; }

}

Thoughts?

alfitec, 2023-09-17:
If you ask me it adds a layer of complexity, I would say that for one device / library / anything one way of addresing is enougth... 

Zero based is better as it covers all with ushort but we should be aware that ushort.MaxValue is noumber of data points-1 as 0 exists as rata point and MaxValue exists as data point... 

For now i have used different address methods only once and it was like ok i just need to add 1... 
So best to keep things simple... but universaly applicable... 
adresses are valid for ushort but all of them...

On the other hand Struct like one above can have implicit cast to ushort or any other integer 
and can be used as parameter without changing anything in current code, so without adding aditional complexity. 

but I do beleive that 1 based addresing is rarely used, only for showing addres with asociated memory area (coil / register ...)

MPapst, 2023-09-17:
I've seen and used a couple of implementations (SCADA, libraries and also PLC or embedded implementations) and - as already said - I see this as an application layer issue. The Protocol implementation should try not to add this complexity. 
Using a 1 based approach translates to a 0 based approach on the transport layer, which will always add confusion and leads to misinterpretation of data - which may not be seen, especially if having a SCADA solution with a lot of signals.

Modbus itself is not a very thankful protocol, because the whole application layer seems to be missing. This makes it very handy on platforms that need a very low footprint and bring interop. But address vs. register and data interpretation is a pain in the ass - always when integrating new devices :)

rquackenbush, 2023-09-18:
You make excellent points.An NModbus user has previously requested some
"equipment specific" nuget packages. This sort of functionality would
probably be a better fit in that sort of area.

alfitec, 2023-09-20:
Solved 65535 problem ... it was +1 looks like its working... 
Now I will try to add test for this if posible... this is much harder for me...
I might need a bit of help ... (I will ask when I get stuck)
I am not realy normaly doing PC programming and this is realy my first time: 
* editing a library
* Writing a test
* colaborating on github
* beeing part of open source project 

Generaly I am controll systems engineer, I write one form aps when i realy need to.... Now a bit more in theory but...

rquackenbush, 2023-09-21:
Looks like this is closed with a PR. Closing for now.

closed 2023-09-21 by rquackenbush