NModbus/NModbus #151: Use RTS line in serial RTU mode

myarc-firmware, 2023-10-02:
Some (in my practice, most) RTU implementations are 2-wire rather than 4. For that reason, most RS-485 adaptor chips have a driver-enable line, which is often tied to the RTS line of the serial port.

There needs to be an option to "use RTS" for communication: toggle RTS high (or low, make it configurable), transmit data, toggle RTS line back to receive.

rquackenbush, 2023-10-02:
Hi - thanks for your interest in NModbus. What is the use case for this
feature?

myarc-firmware, 2023-10-03:
Sorry if I wasn't very clear. The hardware that converts the physical RS-232 voltage levels (or TTL) to RS-485 usually has a Driver Enable (Transmit Enable) pin. It is most commonly connected to the RTS line. RTS does not need to wait for CTS, it is simply used to enable the hardware that outputs the data. This is needed because a lot of hardware uses RS-485 in 2-wire mode (or 3: 2 data liens + ground). In this mode, the same 2 wires (differential pair) are used to transmit and receive data.

When used in 4-wire mode, this is often not needed, it is also not needed when using some USB adaptors, for example, FTDI's USB-RS485-WE-1800-BT. That specific unit is a hard-wired 2-wire RS-485 device which handles enabling transmitter in the hardware, it doesn't care if RTS line is enabled or not, it enables transmitter when sending data, then disables it, so data can be read. However, this is not the universal behaviour or the standard. It is most certainly convenient not to worry about it, but not all hardware is built this way.

If the Transmitter hardware is always enabled in 2-wire mode, then you have a collision when a slave tries to respond. For this reason, RTS line enables the transmitter, data is transmitted, then RTS line disables the transmitter and the unit can receive data from a slave.

This can be an option in the constructor for the serial RTU mode. The default value for `UseRTS` can be false, but the option needs to be there. I should have some free time next week, I'll try to tackle this myself and submit a pull request.

I believe in good faith that this will benefit many people.

rquackenbush, 2023-10-03:
Thank you for the additional information - that's helpful.

Could this be as simple as configuring the [RtsEnable](https://learn.microsoft.com/en-us/dotnet/api/system.io.ports.serialport.rtsenable?view=dotnet-plat-ext-7.0#system-io-ports-serialport-rtsenable) property on the [SerialPort](https://learn.microsoft.com/en-us/dotnet/api/system.io.ports.serialport?view=dotnet-plat-ext-7.0) instance before passing it into [SerialPortAdapter](https://github.com/NModbus/NModbus/blob/master/NModbus.Serial/SerialPortAdapter.cs#L11C18-L11C36)?

```c#
using (SerialPort port = new SerialPort(PrimarySerialPortName))
{
  // configure serial port
  port.BaudRate = 9600;
  port.DataBits = 8;
  port.Parity = Parity.None;
  port.StopBits = StopBits.One;
  port.Open();
  
  //This appears to default to false.
  port.RtsEnable = true;
  
  var factory = new ModbusFactory();
  IModbusMaster master = factory.CreateRtuMaster(port);
  
  byte slaveId = 1;
  ushort startAddress = 100;
  ushort[] registers = new ushort[] { 1, 2, 3 };
  
  // write three registers
  master.WriteMultipleRegisters(slaveId, startAddress, registers);
}
```

rquackenbush, 2023-10-31:
I haven't heard back, so I'm closing this for now. If you don't believe that this resolves the issue, please let me know.

closed 2023-10-31 by rquackenbush