NModbus/NModbus #154: READHOLDINGREGISTERS32 - function is not working right.

newsakthi, 2023-11-02:
I use NMODBUS 3.0.81. I polled a devise (slave) which is configured for 32 bit register. the slave device communicates properly.  but I get an error "unexpected byte count Expected 4 but received 8". 
My query is ReadHoldingRegisters32(1,7001,1). it sends the byte array "01 03 1B 59 00 02 12 FC". 
I receive the reply from the slave devise as "01 03 08 47 31 30 00 00 00 00 00 F5 32". (The value stored in 7001 register is 45360 and 7002 is 0).
Please help to resolve the issue.

rquackenbush, 2023-11-03:
Hi - the request for ReadHoldingRegisters32 is implemented as an extension method that just doubles the number of registers requested from the slave:

```c#
public static uint[] ReadHoldingRegisters32(this IModbusMaster master, byte slaveAddress, ushort startAddress, ushort numberOfPoints)
{
    var registers = master.ReadHoldingRegisters(slaveAddress, startAddress, (ushort)(numberOfPoints * 2));

    return ConvertTo32(registers);
}
```

For the call `ReadHoldingRegisters32(1,7001,1)`, that gives us:

```c#
var registers = master.ReadHoldingRegisters(1, 7001, 2);
```

Given the spec for "Read Holding Registers"
[image]

That should give us the following request (not including the two CRC bytes):

| Hex | Decimal | Size | Description |
|--|--|--|--|
| `0x01`     | 1       | 1 byte    | Slave Id | 
| `0x03`     |  1      | 1 byte   | Function Code - Read Multiple Holding Registers |
| `0x1B59` | 7001 | 2 bytes  | Starting Address |
| `0x0002` | 2       | 2 bytes  | Quantity of Registers |
| `0x????`  | ?      | 2 bytes  | CRC |

|  | Value |
| --- | --- |
| Expected  |  `01 03 1B 59 00 02 ?? ??` |
| Observed |  `01 03 1B 59 00 02 12 FC` |

Or:  which appears to match the request you provided.

The response should look like:

| Hex | Decimal | Size | Description |
|--|--|--|--|
| `0x01`     | 1       | 1 byte    | Slave Id | 
| `0x03`     | 1       | 1 byte   | Function Code - Read Multiple Holding Registers |
| `0x04`     | 4       | 1 byte  | Byte Count |
| `0xB130` | 45360 | 2 bytes  | Registers[0] |
| `0x0000` | 0       | 2 bytes  | Registers[1] |
|  `0x????` | ?      | 2 bytes  | CRC |

|  | Value |
| --- | --- |
| Expected          | `01 03 04 B1 30 00 00 ?? ??` |
| Observed         | `01 03 08 47 31 30 00 00 00 00 00 F5 32` |

# Initial Assessment

According to the message frames you provided, it appears that the slave device is responding incorrectly.  However, you stated that the same call works in version 3.0.78, which is flummoxing. I _did_ recently change the implementation of `ReadHoldingRegisters32`, but it _theoretically_ should be functionally equivalent. I have unit testing coverage over this function and it _appears_ to be working. This is flummoxing.

I'm out of time for the moment, but when I circle back I'll compare to the execution of the 3.0.78 implementation.

**Question:** What device (manufacturer / model) are you talking to?

newsakthi, 2024-08-31:
it is for SCADA. the slave devices are from honeywell, emerson etc.,

FHuijsmans, 2025-04-28:
Dear newsakthi,
I'm encountering the same problem! However, you mentioned the fact that it's working fine now. Could you inform me about the solution for this issue.

Kind regards