NModbus/NModbus #155: Retries do not work with UdpClient

Jeroen6, 2023-11-02:
It looks like the retries do not work when using an UdpClient.
UdpClient throws an SocketError since it throws a SocketException when there is a read timeout of the acknowledge due to packet loss.

```
   at System.Net.Sockets.Socket.Receive(Byte[] buffer)
   at NModbus.IO.UdpClientAdapter.Read(Byte[] buffer, Int32 offset, Int32 count) in ...NModbus-3.0.80\NModbus\IO\UdpClientAdapter.cs:line 87
   at NModbus.IO.ModbusIpTransport.ReadRequestResponse(IStreamResource streamResource, IModbusLogger logger) in ...NModbus-3.0.80\NModbus\IO\ModbusIpTransport.cs:line 36
   at NModbus.IO.ModbusIpTransport.ReadResponse[T]() in ...NModbus-3.0.80\NModbus\IO\ModbusIpTransport.cs:line 142
   at NModbus.IO.ModbusTransport.UnicastMessage[T](IModbusMessage message) in ...NModbus-3.0.80\NModbus\IO\ModbusTransport.cs:line 137
```
```
A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
```

Right now this walks falls through the retries to the throw at line 198.
https://github.com/NModbus/NModbus/blob/a1d9b8ced766463abbb9997039cee687582887a9/NModbus/IO/ModbusTransport.cs#L205C31-L205C31

I can make it work with an additional `e is SocketException` in the if() at L189

https://github.com/NModbus/NModbus/blob/a1d9b8ced766463abbb9997039cee687582887a9/NModbus/IO/ModbusTransport.cs#L189

rquackenbush, 2023-11-03:
The short answer is - if it works, it works. :-)

Are you in a position to submit a PR?

Jeroen6, 2023-11-08:
Yes, I'll work on that.