NModbus/NModbus #161: Modbus TCP to Serial Converter

Zapnologica, 2023-11-28:
Good day,

i wanted to find out,  does NModbus have the ability to listen as Modbus TCP server, and then route the messages it receives to Serial / Comport and then relay the response and or error to the modbus TCP?

MPapst, 2023-11-28:
You would need to implement that connection on your own.

A short how to in terms of NModbus:
1. implement some kind of interaction service, which offers a method to be called from the server, that can forward data to your serial modbus client.
2. implement your data store that allows to forward updates when there is something written on your modbus server.
Server Implementation Sample: https://github.com/NModbus/NModbus/blob/13a454bdc8dd4b5c0399090a5544927235046a65/Samples/Program.cs#L552
DataStorage sample:
https://github.com/NModbus/NModbus/blob/master/Samples/SparsePointStorage.cs
3. implement the modbus serial client
e.g. from this sample:
https://github.com/NModbus/NModbus/blob/13a454bdc8dd4b5c0399090a5544927235046a65/Samples/Program.cs#L68

Zapnologica, 2023-12-05:
Thank you for this valuable information.

Im not sure if I'm being daft, however I cant seem to find where SparsePointStorage is being used?

Can you please point me to the documentation where you link the SparsePointStorage up to the Modbus TCP slave?  I can't seem to find an example of that?

MPapst, 2023-12-05:
`SparsePointStorage` is an implementation of `IPointSource` which is needed in an implementation of `ISlaveDataStore` (as seen in the same file, just above).

The SlaveNetwork constructor allows to hand over an own instance of an `ISlaveDataStore`. This is also in the examples:
https://github.com/NModbus/NModbus/blob/13a454bdc8dd4b5c0399090a5544927235046a65/Samples/Program.cs#L434C44-L434C44

closed 2023-12-06 by Zapnologica