NModbus/NModbus #164: Exception: The requested address is not valid in its context

kelilive, 2023-12-19:
Communicating via Modbus RTU over TCP/IP Protocol. I don't know why it went wrong. Can you guide me?

[image]
[image]

Stack Trace:
```
at System.Net.Sockets.Socket.DoBind(EndPoint endPointSnapshot, SocketAddress socketAddress)
at System.Net.Sockets.Socket.Bind(EndPoint localEP)
at System.Net.Sockets.TcpListener.Start(Int32 backlog)
```

MPapst, 2023-12-19:
A TCPListener listenes on a given IP address of the local computer.

So, if you are trying to listen on an ip, the current computer needs an interface with that ip assigned to it.

If you are connecting from and to the same pc, you can use the loopback address. If not on the same pc, you can use the "any" address.
Both can be used from static properties of the IPAddress class (IPAddress.Any).

kelilive, 2023-12-20:
These two addresses are both LAN addresses, although they are not very similar. They can be pinged normally and all firewalls have been turned off. Because I can read data normally using Modbus Poll, but I don't know how to implement it using NModbus. My local IP is 196.196.198.152.

Can you elaborate on what the code should do? Grateful!

MPapst, 2023-12-20:
If 196.196.198.152 is your local IP of your computer - the computer is directly connected to the internet without any firewall/router?

Try: 
```csharp
TcpListener listener = new TcpListener(IPAddress.Any);
```

kelilive, 2023-12-21:
[image]

```csharp
public static ushort[] ReadHoldingRegisters()
{
    const byte slaveId = 1;
    const int port = 20108;

    var listener = new TcpListener(IPAddress.Any, port);

    listener.Start();

    var factory = new ModbusFactory();
    var network = factory.CreateSlaveNetwork(listener);

    var slave = factory.CreateSlave(slaveId);

    network.AddSlave(slave);
    network.ListenAsync();

    var address = new IPAddress(new[] { (byte)196, (byte)196, (byte)196, (byte)49 });
    using var client = new TcpClient(new IPEndPoint(address, port));
    var master = factory.CreateMaster(client);

    return master.ReadHoldingRegisters(0, 52, 2);
}
```

MPapst, 2023-12-21:
Have a look in the documentation. There is a difference in creating the TcpClient while using an IPEndPoint or using a string for the IP/using the ConnectAsync method.
https://learn.microsoft.com/en-us/dotnet/api/system.net.sockets.tcpclient?view=net-8.0#constructors

kelilive, 2023-12-21:
I have basically mastered the usage of tcp, although I don't use it much, the problem now is that the interface is not working.

kelilive, 2023-12-21:
```csharp
RawModbusRtuOverTcp.Test1();

Console.WriteLine("--------");
RawModbusRtuOverTcp.Test2(1, 1);

Console.WriteLine("--------");
RawModbusRtuOverTcp.Test3(1, 1);

Console.WriteLine("--------");
RawModbusRtuOverTcp.Test4(1, 1);
```

I tried a more primitive approach, but I don't know how to parse this data that returns normal int, float, double. The following is a description of the problems I have encountered. I hope some warm-hearted friends can give me some help.

When I run it for the first time, `Test1`, `Test2`, and `Test3` can get the data normally, and reading a simple length of data is the same result. However, if `Test4` cannot be obtained, it will be stuck at the time of Receive and cannot proceed to the next line. I have slightly improved the code. I don't know what the problem is. Now only the first one can return the result, and the other 3 can't return the data.

[RawModbusRtuOverTcp.zip](https://github.com/JamesNK/Newtonsoft.Json/files/13741940/RawModbusRtuOverTcp.zip)

kelilive, 2023-12-21:
```csharp
public static void Main()
{
    var client = new TcpClient();

    client.Connect("196.196.196.49", 20108);

    var factory = new ModbusFactory();
    var master = factory.CreateMaster(client);
    var registers = master.ReadHoldingRegisters(1, 52, 2);
    var data = ModbusUtility.GetSingle(registers[0], registers[1]);

    Console.WriteLine(data);
}
```

[image]

kelilive, 2023-12-22:
[TestDemo.zip](https://github.com/NModbus/NModbus/files/13748167/TestDemo.zip)

MPapst, 2023-12-22:
NModus takes the Modbus standard as is.

What you can do is to capture the traffic using Wireshark and see if Wireshark can decode the traffic as Modbus properly or not.

Magniveo, 2023-12-22:
[image]
I used this library more than 4 years.

closed 2023-12-25 by kelilive

kelilive, 2023-12-25:
Later, on the one hand, I deepened my understanding, and on the other hand, I used more scientific methods to analyze and investigate. Including the research on packet capture software you mentioned. I no longer do tests that have poor control perception. I virtualized the serial port myself using virtual serial port software. Then Modbus Slave and Modbus Poll easily passed various tests, except for the ASCII protocol. Then cooperate with ILSPY to find matching classes and methods, and finally the test passes. After figuring out how to make the code, replacing the IP and port of the serial port server was successfully completed. Failure to pass the ModBus test for the TCP standard protocol means that the configuration cannot communicate normally. It is almost not easy to communicate when I send commands myself, so I can understand and accept them.

```
public static float ReadHoldingRegisters(string host, int port, ushort startAddress, ushort length)
{
    using var client = new TcpClient();

    client.Connect(host, port);

    var factory = new ModbusFactory();
    var adapter = new TcpClientAdapter(client);
    var master = factory.CreateRtuMaster(adapter);
    var registers = master.ReadHoldingRegisters(1, startAddress, length);

    return ModbusUtility.GetSingle(registers[0], registers[1]);
}
```

[image]
[image]

weirdyang, 2024-12-05:
> ![2023-12-22 (1)](https://private-user-images.githubusercontent.com/5388016/292560589-ca178966-51ee-4e41-965b-0ea16754c018.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0MDAwNjMsIm5iZiI6MTczMzM5OTc2MywicGF0aCI6Ii81Mzg4MDE2LzI5MjU2MDU4OS1jYTE3ODk2Ni01MWVlLTRlNDEtOTY1Yi0wZWExNjc1NGMwMTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDVUMTE1NjAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmQxNWJmZGQ1MWNjMmFkNTJmZjZjOTgwZmFiMDFjOWM4YjgxMDE5Y2E2NDIyMDU4MzNlYzE1ZWE0MjA4NDkxZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.QtmOjQUvgqSAUU62gd1c0hTq_0ROw3omF2by-6ybqDY) I used this library more than 4 years, all protocols work.

is this project available for reference?

Magniveo, 2024-12-06:
> is this project available for reference?

Im writed project for self, with stack electrondotnet and blazor he have very more error. May be in future i will port him to the MAUI and then will publish. May be will interest this project.
https://github.com/AndreyAbdulkayumov/TerminalProgram