NModbus/NModbus #166: ReadWriteMultipleRegistersService

markus-thomi-condair, 2024-01-05:
For an old modbus legacy device I need the method: ReadWriteMultipleRegisters()
In the current implementation this function throws a NotSupportedException() in ReadWriteMultipleRegistersService.cs
**NModbus\Device\MessageHandlers\ReadWriteMultipleRegistersService.cs**

In order to make it work, it is enough to fix this method GetRtuResponseBytesToRead as followed:

        public override int GetRtuResponseBytesToRead(byte[] frameStart)
        {
            //throw new NotSupportedException();
            return frameStart[2] + 1;
        }

With this fix I could use the ReadWriteMultipleRegisters() successfully.

Magniveo, 2024-01-16:
It is better to do it via the `NModbus.IModbusMessage` interface, with request and response. You can see examples of this in the documentation.

markus-thomi-condair, 2024-04-03:
Thank you for the hint. I use the ModbusFactory with a Customized ReadWriteMultipleRegistersService implementation:

```
public class CustomReadWriteMultipleRegistersService : ReadWriteMultipleRegistersService
 {
     public override int GetRtuResponseBytesToRead(byte[] frameStart)
     {
         return frameStart[2] + 1;
     }
 }

  private IModbusMaster CreateModbusRtuMaster(SabModbusRtuConfig config)
 {

     var functionServices = new IModbusFunctionService[]
     {
             new CustomReadWriteMultipleRegistersService()
     };

     var factory = new ModbusFactory(functionServices);
     ....
   }
```