NModbus/NModbus #167: System.IO.IOException: Response was not of expected transaction ID. Expected 1637, received 1636.

zhangxize, 2024-01-11:
System.IO.IOException: Response was not of expected transaction ID. Expected 1637, received 1636.
   at NModbus.IO.ModbusIpTransport.OnValidateResponse(IModbusMessage request, IModbusMessage response)
   at NModbus.IO.ModbusTransport.ValidateResponse(IModbusMessage request, IModbusMessage response)
   at NModbus.IO.ModbusTransport.UnicastMessage[T](IModbusMessage message)
   at NModbus.Device.ModbusMaster.PerformReadDiscretes(ReadCoilsInputsRequest request)
   at Communication.Core.ModbusHelper.ReadInputs(UInt16 startAddress, UInt16 numberOfPoint)
   at Communication.Core.MainControlService.Heartbeat()

NModbus connect modbus-slave tools ,Occasionally the exception.

MPapst, 2024-01-11:
This might be an issue with the device you are communicating with. It is not responding with the correct transaction id.

elcdrue, 2024-03-05:
I have simular issue. I captire traffic with wireshark. 
Dump file have TCP Spurious Retransmission of previous response frame.

Rofes, 2024-04-04:
I have this issue when communicating modbus over WIFI. It's easy to reproduce:
1. Send query (transaction id=0)
2. Received data
3. Block WIFI signal of server device
4. Send another query (transaction id=1)
5. Wait for IOException on client side
6.  Send another query (transaction id=2)
7. Unblock WIFI signal of server device
8. Server will recieve query requests of transaction 1 and 2.
9. Server sends response on transaction 1 and 2.
10. NModbus will process transaction 1 while it is expecting transaction 2.

It seems that 2 response messages are buffered, but only first message is processed.
I should expect that readbuffer will be read till correct transaction is found or till buffer is completely read out.

Rofes, 2024-04-05:
I have solved this ussue using:
mMaster.Transport.RetryOnOldResponseThreshold = 10;

However, it seems not to be a correct implementation to solve this issue.

closed 2024-04-22 by zhangxize