NModbus/NModbus #172: Is Nodbus thread safe with serial ports?

Zapnologica, 2024-01-31:


Magniveo, 2024-02-01:
Yes, BlockingQueue will be a nice solution.
And you can use master.ReadHoldingRegistersAsync.

Zapnologica, 2024-02-01:
Sorry Just to clarify.

Yes, it is inherently thread-safe?

So I can have 30 threads all calling  `master.ReadHoldingRegistersAsync.` at once, and the library will sequentiallyize them and poll one at a time.

Magniveo, 2024-02-01:
Not tested with more than 10 and need to know how quick a device can answer because devices need delay.

Zapnologica, 2024-02-06:
Ok this is what I am trying to establish. Do I need to manage this backoff and retry or will the library handle that?  I don't want to re-invent the wheel.

Unfortunately in my case I don't have any control over how many and how fast the requests will come.  So I need to ensure the serial port is protected.

The retry value and wait between retires.   Does the nmodbus library implement this further down the food chain> I cant find any good documentation on it..

So for example if we do a read register, and it times out.  (if I have retry set to 3). Will it throw a timeout exception on the first fail. Or only after all 3 retires have happened?