NModbus/NModbus #173: Port.Read Timeout on a SerialPort

Zapnologica, 2024-02-01:
Good day,

I would like to understand how the library works with serial timeouts?

I have the following test code:

`  var serialPort = new SerialPort("COM1", 9600, Parity.None, 8, StopBits.One);

  //Set Serial Timeouts
  serialPort.ReadTimeout = 500;
  serialPort.WriteTimeout = 100;

  var factory = new ModbusFactory();
  IModbusMaster master = factory.CreateRtuMaster(serialPort);

  //Change settings
  master.Transport.Retries = 0;
  master.Transport.ReadTimeout = 1000;
  master.Transport.WriteTimeout = 200;

  //Read device which is offline.
  master.ReadHoldingRegisters(1, 1, 1);`

Which timeout values will be applied to the actual physical rs485 serial port?

A: 550 & 100   
or
B: 1000 & 200

Magniveo, 2024-02-01:
You have rewritten its settings here.
```
master.Transport.ReadTimeout = 1000;
master.Transport.WriteTimeout = 200;
```

Zapnologica, 2024-02-01:
OK,  so those values are just a higher layer abstraction of the actual serial port values.

That makes sense.

rquackenbush, 2024-02-01:
The last value will win.

The line:
master.Transport.ReadTimeout = 1000

Just calls serialPort.ReadTimeout under the covers.