NModbus/NModbus #174: How to tell slave to not send data until data is save to send?

CodeMarc42, 2024-02-08:
Hi, i have a slave that i have to write in them coils and registers, then i do it first writing coils, and then writing the registers.

But what will happend if a master request me data, if i dont had writted all?  My idea was to try to implement a event when a request to the slave arribe, to then in there make it wait until some flag tells the program that the new data is fully writed and updated....

But reading and searching for info about this i m not lucky and not found a tip yet on how to do it.

Any idea, example ot method that i m missing and that can help me in that issue? Thanks

CodeMarc42, 2024-02-08:
After searching more and keep doing test:

        SlaveDataStore dataStore = new NModbus.Data.SlaveDataStore();
        dataStore.CoilDiscretes.BeforeRead += OnBeforeRead;
        this.slave = factory.CreateSlave(1, dataStore);

You think that something like this will do the trick? Placing in OnBeforeRead a flag or mutex to know when the data that can i be writing updated?

1 WriteCoils, 
2 ReadCoils (check updating flag or mutex, cause i m yet updating data i make the read to wait)
3 WriteREgisters
4 Setup flag or mutex, updating data to false
5 Previos ReadCoils continue and end the read and send data to requester

Magniveo, 2024-02-09:
I might be wrong, you can make it via functionService, look in the sample. 
`StartModbusSerialRtuSlaveWithCustomMessage`