NModbus/NModbus #185: Response was not of expected transaction ID. Expected 4, received 1.

Zapnologica, 2024-05-17:
Good day,

I am using Nmodbus to read modbus tcp, but from another computer also running Nmodbus as a slave?

I seem to get a lot of errors like this:

"[12:54] Brandon Sutherland
Response was not of expected transaction ID. Expected 4, received 1."

Any ideas what could causing this?

jimlojh, 2024-06-07:
I also encountered this problem and solved it. The reason is that I found that there is a certain probability that two different threads or schedules accidentally call IModbusMaster (global variable).ReadHoldingRegisters at the same time. The read length parameters of the two calls are different, which will cause the subsequent reading to be completely disordered, and the expectation and return are inconsistent.
I analyzed the request and reply input parameters of the synchronous reading method of the Nmodbus source code and suspected that it had a bug. Later, I changed to use the asynchronous reading method of Nmodbus (await modbusMaster.ReadHoldingRegistersAsync), and the test can work normally.
[image]

Zapnologica, 2024-06-07:
To my knowledge, this library is meant to be thread safe?

In our application, it will most certainly have multiple threads trying to read registers at varying different times.  Surely the library just queues them and polls them one at a  time? Or times out?

jimlojh, 2024-06-17:
Most likely it is caused by timeout.
[image]
[image]
You can set the RetryOnOldResponseThreshold (default = 1) to make the transaction ID correct.

Zapnologica, 2024-10-03:
I am still struggling with this issue, its causing major havock.

I have reworked our entire application so that it queues all concurrent requests and I think have a single threat that is making the requests to nmobus library.  So I have an instance on Nmodbus per device and a single thread per device.

I am talking to devices on a local LAN, there should be NO timeouts happening..  And I get this issue fairly regularly..

I am worried that its a bug in this library and there isn't actually a timeout?