NModbus/NModbus #190: Visual Studio Build Errror ,Code [CS1503] Failed  to  trans   [System.IO.Ports.SerialPort] to  [NModbus.IO.IStreamResource]

szh8052, 2024-07-16:
Sorry for my pool english , and thank for the author to publish that gread project.
AS the picture follow, i try to create a modbus_rtu master with serial com10 . But when i try to build the project , Visual Studio tell me something error . 
[image]
It's very sad to seen this. I copy the code from sample and uninstall  and reinstall  "System.IO.Ports" , "NModbus" with nuget but it not work.
Here are some information about nuget bags 。
[1] System.IO.Ports Version : 8.0.0
[2] NModbus Version:  3.0.81

My Visual Studio Version:  Microsoft Visual Studio Community 2022 (64 位) - Current 17.7.5

Project sample code:
[image]
[link[https://github.com/NModbus/NModbus/blob/master/Samples/Program.cs]](https://github.com/NModbus/NModbus/blob/master/Samples/Program.cs)

MPapst, 2024-07-17:
Same issue as #159

You need to install the Serial Port package: NModbus.Serial

Conn-Ho, 2024-07-18:
> Same issue as #159
> 
> You need to install the Serial Port package: NModbus.Serial, that adds the necessary Extension method.

Thank you  That's help me a lot