NModbus/NModbus #196: Heart Beat Funtion to know Server service closed

Lj-dog, 2024-09-06:
I google it, found all results is that build a thread to read a PLC register and "try-catch" it waiting to throw exception. Does NModbus have function like this so I could follow DRY(Don't Repeat Yourself) principle. I want to find a better way like event handler or trigger.
  Thanks for your reply.

closed 2024-12-18 by Lj-dog

cyrine3110, 2025-02-11:
Hello, i'm encountering the same problem, can you tell me how you solved this ?

Lj-dog, 2025-02-11:
> Hello, i'm encountering the same problem, can you tell me how you solved this ?

Like I said.  I start a "Task" function like "private async Task CheckConnect(CancellationToken token,Action CallBack) when I connected PLC and don't "await" it. 

```
private CancellationTokenSource HeartBeatCTS {get;set;}

pubulic bool PLCConnect(...)
{
    //your connect operation
    //if connnected,start Heart Beat Check
    if(connect.IsSuccess)
    {
        HeartBeatCTS = new CancellationTokenSource();
        CheckConnect(HeartBeatCTS.Token,CallBack);
    }
}
```

So that the Heart Beat can run in background. Until you don't need it , use "HeartBeatCTS.Cancel()" to stop it.

in function CheckConnect(..) , you should read something  continually  , until it lose connection or you don't need it.

```
  private async Task CheckConnect(cancellationToken token,Action Callback)
{
    try
    {
        while(true)
        {
            token.ThrowIfCancellationRequested();
            await res = ReadAsync(plcAddress)
            if(!res)
             {

                Callback();
                // disconnect PLC
             }
             await Task.Delay(500); //Heart Beat time interval

        }
    }
    catch
    {

    }
}
```